braidstag/arduino-milestag #27: 3D Print prototype enclosures

neothermic, 2018-07-19:
Small prototyping enclosures large enough to contain electronics and battery, these don't particularly need to be gun shaped, or very ergonomic, and would ideally include the lens optics aswell.

Note varying/changing size and shape of electronic components during design process

sbumstead, 2018-09-09:
I can assist in creating the development enclosures. Sean

bumsteadsean@gmail.com

sbumstead, 2018-09-09:
I have experience designing and printing custom laser tag guns.

ashirley, 2018-09-09:
The real enclosure will be based on the P90 assault rifle but I have created a pistol as a quick way to carry all the electronics around while developing. The pistol is not perfect (it is probaly too fragile for real play, the edges are too sharp and the microswitch for the trigger is too difficult to install) but it is good enough for developing.

You can see the current pistol design on https://a360.co/2x14Sf0