braidstag/arduino-milestag #42: Identity and status lights

neothermic, 2019-08-16:
Should we have status LEDs that are always or periodically visible to indicate team ID or other effects such as invulnerability, quaddamage, dead, etc?

These would have an impact on power consumption, immersion, stealth etc so would probably need to be optional. 

Potentially opens up possibilities of "reveal enemy" mechanics (see stompyrobots "radar hack")

Related to #38

neothermic, 2019-08-16:
This would likely at this point require an extra driver board, or external LED driver package which would need controlling somehow. At a glance we're either looking at a second Atmega chip running in tandem  (could this share the same serial connection to the pi as the first, with each chip ignoring commands not destined for it?) or driving LEDs directly from the pi using some other form of driver.

ashirley, 2019-08-18:
the serial connection is point to point (so can't have multiple devices unlike I2C/SPI) so we would have something else. This isn't a deal breaker though and we might just be able to control neopixels from the pi directly.